Bumblebee-Project/bbswitch #198: bbswitch wont compile vs 5.6.3

rubiksdot, 2020-04-09:
A little bleeding edge but it would appear that bbswitch no longer compiles vs kernel 5.6.3. This is the end of the make log that I have:

```
  CC [M]  /var/lib/dkms/bbswitch/0.8/build/bbswitch.o
/var/lib/dkms/bbswitch/0.8/build/bbswitch.c: In function ‘bbswitch_init’:
/var/lib/dkms/bbswitch/0.8/build/bbswitch.c:460:63: error: passing argument 4 of ‘proc_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
     acpi_entry = proc_create("bbswitch", 0664, acpi_root_dir, &bbswitch_fops);
                                                               ^~~~~~~~~~~~~~
In file included from ./include/acpi/acpi_bus.h:83,
                 from ./include/linux/acpi.h:32,
                 from /var/lib/dkms/bbswitch/0.8/build/bbswitch.c:32:
./include/linux/proc_fs.h:64:24: note: expected ‘const struct proc_ops *’ but argument is of type ‘struct file_operations *’
 struct proc_dir_entry *proc_create(const char *name, umode_t mode, struct proc_dir_entry *parent, const struct proc_ops *proc_ops);
                        ^~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:268: /var/lib/dkms/bbswitch/0.8/build/bbswitch.o] Error 1
```

I'm guessing that changes made to use file descriptors for processes are causing the issue.

ArchangeGabriel, 2020-04-10:
#196

ohquait, 2020-04-11:
just a test but for me works:
[bbswitch-v0.8-proc_ops-struct.patch](https://www.qua-it.org/XFCE/shots/share-with-pclos/pclos/bbswitch-v0.8-proc_ops-struct.patch)

now it compiles good

Lekensteyn, 2020-05-19:
Fix merged with #196

closed 2020-05-19 by Lekensteyn

rien333, 2020-05-26:
I'm testing out 5.7 (to be released next week, I think?) because I have a different problem. Now, I found out that bbswitch  doesn't compile, again related to stuff with proc_ops. See the included file for a build log.

[make.log](https://github.com/Bumblebee-Project/bbswitch/files/4683433/make.log)

Lekensteyn, 2020-05-26:
@rien333 Please try the latest develop branch, an implicit header was removed by upstream so we have to explicitly include it.

rien333, 2020-05-26:
Works again, thank you very much!