Bumblebee-Project/bbswitch #199: bbswitch.c fails to compile under kernel 5.6+

offsides, 2020-05-28:
Because of a change in the argument to proc_create, bbswitch.c fails to compile under kernel 5.6 or later.  Here's the quick-and-dirty change I made to fix it, but it should probably be done as a cleaner, conditional patch checking the kernel version.

```
--- bbswitch.c.orig     2019-03-07 19:37:51.000000000 -0500
+++ bbswitch.c  2020-05-28 14:04:16.905359570 -0400
@@ -375,12 +375,12 @@
     return 0;
 }
 
-static struct file_operations bbswitch_fops = {
-    .open   = bbswitch_proc_open,
-    .read   = seq_read,
-    .write  = bbswitch_proc_write,
-    .llseek = seq_lseek,
-    .release= single_release
+static struct proc_ops bbswitch_fops = {
+    .proc_open    = bbswitch_proc_open,
+    .proc_read    = seq_read,
+    .proc_write   = bbswitch_proc_write,
+    .proc_lseek   = seq_lseek,
+    .proc_release = single_release
 };
 
 static struct notifier_block nb = {
```

ArchangeGabriel, 2020-05-28:
https://github.com/Bumblebee-Project/bbswitch/commit/b0fdcfd847ecf5cbe6754b50f0db78600380b9f0 and https://github.com/Bumblebee-Project/bbswitch/commit/ddbd243638c7bc2baecf43a78aff46cdc12e9b2e

closed 2020-05-28 by ArchangeGabriel

offsides, 2020-05-28:
Thanks!  Looks like a case of concurrent timing. :)