Bumblebee-Project/bbswitch #205: F31: kernel 5.8.8 build fails for Bumblebee

Cyborgscode, 2020-09-15:
OS: Fedora 31+
Short: Bug in BB sourcecode 

Results in:

  Ausgeführtes Scriptlet: kernel-core-5.8.8-100.fc31.x86_64                                                                                                                                                                                                               32/32 
dkms: running auto installation service for kernel 5.8.8-100.fc31.x86_64

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.8.8-100.fc31.x86_64 KVERSION=5.8.8-100.fc31.x86_64....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.8-100.fc31.x86_64 (x86_64)
Consult /var/lib/dkms/bbswitch/0.8.0/build/make.log for more information.
 Done. 
dkms: running auto installation service for kernel 5.8.8-100.fc31.x86_64

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.8.8-100.fc31.x86_64 KVERSION=5.8.8-100.fc31.x86_64....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.8-100.fc31.x86_64 (x86_64)
Consult /var/lib/dkms/bbswitch/0.8.0/build/make.log for more information.
 Done. 

  Ausgeführtes Scriptlet: armadillo-9.900.2-1.fc31.x86_64                                             

Because of:

# cat /var/lib/dkms/bbswitch/0.8.0/build/make.log
DKMS make.log for bbswitch-0.8.0 for kernel 5.8.8-100.fc31.x86_64 (x86_64)
Di 15. Sep 11:33:46 CEST 2020
make -C /lib/modules/5.8.8-100.fc31.x86_64/build M="$(pwd)" modules
make[1]: Verzeichnis „/usr/src/kernels/5.8.8-100.fc31.x86_64“ wird betreten
  CC [M]  /var/lib/dkms/bbswitch/0.8.0/build/bbswitch.o
/var/lib/dkms/bbswitch/0.8.0/build/bbswitch.c: In Funktion »bbswitch_init«:
/var/lib/dkms/bbswitch/0.8.0/build/bbswitch.c:460:18: Fehler: Implizite Deklaration der Funktion »proc_create« [-Werror=implicit-function-declaration]
  460 |     acpi_entry = proc_create("bbswitch", 0664, acpi_root_dir, &bbswitch_fops);
      |                  ^~~~~~~~~~~
/var/lib/dkms/bbswitch/0.8.0/build/bbswitch.c:460:16: Warnung: Zuweisung an »struct proc_dir_entry *« von »int« wandelt eine Zahl in einen Zeiger um, ohne explizite Typkonvertierung [-Wint-conversion]
  460 |     acpi_entry = proc_create("bbswitch", 0664, acpi_root_dir, &bbswitch_fops);
      |                ^
/var/lib/dkms/bbswitch/0.8.0/build/bbswitch.c: In Funktion »bbswitch_exit«:
/var/lib/dkms/bbswitch/0.8.0/build/bbswitch.c:490:5: Fehler: Implizite Deklaration der Funktion »remove_proc_entry« [-Werror=implicit-function-declaration]
  490 |     remove_proc_entry("bbswitch", acpi_root_dir);
      |     ^~~~~~~~~~~~~~~~~
cc1: Einige Warnungen werden als Fehler behandelt
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/bbswitch/0.8.0/build/bbswitch.o] Fehler 1
make[1]: *** [Makefile:1756: /var/lib/dkms/bbswitch/0.8.0/build] Fehler 2
make[1]: Verzeichnis „/usr/src/kernels/5.8.8-100.fc31.x86_64“ wird verlassen
make: *** [Makefile:13: default] Fehler 2

Ambrevar, 2020-09-27:
Same error (in English) with Linux 5.8.11:

```
make: Entering directory '/gnu/store/jghf3pd73jpdf6balqgjllaqccvh4rpz-linux-libre-module-builder-5.8.11/lib/modules/build'
  AR      /tmp/guix-build-bbswitch-module-0.8.drv-0/source/built-in.a
  CC [M]  /tmp/guix-build-bbswitch-module-0.8.drv-0/source/bbswitch.o
/tmp/guix-build-bbswitch-module-0.8.drv-0/source/bbswitch.c: In function 'bbswitch_init':
/tmp/guix-build-bbswitch-module-0.8.drv-0/source/bbswitch.c:460:18: error: implicit declaration of function 'proc_create'; did you mean 'vfs_create'? [-Werror=implicit-function-declaration]
     acpi_entry = proc_create("bbswitch", 0664, acpi_root_dir, &bbswitch_fops);
                  ^~~~~~~~~~~
                  vfs_create
/tmp/guix-build-bbswitch-module-0.8.drv-0/source/bbswitch.c:460:16: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     acpi_entry = proc_create("bbswitch", 0664, acpi_root_dir, &bbswitch_fops);
                ^
/tmp/guix-build-bbswitch-module-0.8.drv-0/source/bbswitch.c: In function 'bbswitch_exit':
/tmp/guix-build-bbswitch-module-0.8.drv-0/source/bbswitch.c:490:5: error: implicit declaration of function 'remove_proc_entry'; did you mean 'remove_memory'? [-Werror=implicit-function-declaration]
     remove_proc_entry("bbswitch", acpi_root_dir);
     ^~~~~~~~~~~~~~~~~
     remove_memory
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:281: /tmp/guix-build-bbswitch-module-0.8.drv-0/source/bbswitch.o] Error 1
make: *** [Makefile:1752: /tmp/guix-build-bbswitch-module-0.8.drv-0/source] Error 2
```

gsgatlin, 2020-09-27:
I think you are running into this perhaps

https://github.com/Bumblebee-Project/bbswitch/pull/196

So this patch should work around it or you could use the upstream version.

https://pastebin.com/2D4ewzC2

So I guess you need to ask the person whop built this for fedora to add that patch to it or git clone the newest version.

Ambrevar, 2020-09-27:
It works indeed, thanks!

I've updated the Guix package to follow the "develop" branch, it works
now.