Bumblebee-Project/bbswitch #206: Can't shutdown card

ouafnico, 2020-10-30:
Hello

I'm using a lenovo t490 on archlinux with these specs : 
- GeForce MX250
- intel gpu on i7-8665U

I'm trying to kill the card with bbswitch but it failed

[ven. oct. 30 11:16:02 2020] bbswitch: version 0.8
[ven. oct. 30 11:16:02 2020] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
**[ven. oct. 30 11:16:02 2020] bbswitch: Found discrete VGA device 0000:3c:00.0: \_SB_.PCI0.RP09.PEGP**
[ven. oct. 30 11:16:02 2020] ACPI Warning: \_SB.PCI0.RP09.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20190816/nsarguments-59)
[ven. oct. 30 11:16:02 2020] bbswitch: detected an Optimus _DSM function
[ven. oct. 30 11:16:02 2020] pci 0000:3c:00.0: enabling device (0006 -> 0007)
**[ven. oct. 30 11:16:02 2020] bbswitch: Succesfully loaded. Discrete card 0000:3c:00.0 is on**
[ven. oct. 30 11:16:15 2020] bbswitch: disabling discrete graphics
**[ven. oct. 30 11:16:42 2020] bbswitch: disabling discrete graphics**
[ven. oct. 30 11:16:42 2020] ------------[ cut here ]------------
**[ven. oct. 30 11:16:42 2020] pci 0000:3c:00.0: disabling already-disabled device**
[ven. oct. 30 11:16:42 2020] WARNING: CPU: 2 PID: 6751 at drivers/pci/pci.c:1942 pci_disable_device+0x9e/0xc0
[ven. oct. 30 11:16:42 2020] Modules linked in: bbswitch(OE) tun snd_seq_dummy snd_hrtimer snd_seq snd_seq_device fuse xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc ccm joydev mousedev overlay iTCO_wdt iTCO_vendor_support snd_hda_codec_hdmi mei_wdt mei_hdcp intel_rapl_msr wmi_bmof intel_wmi_thunderbolt x86_pkg_temp_thermal intel_powerclamp coretemp nls_iso8859_1 kvm_intel snd_sof_pci nls_cp437 snd_sof_intel_byt snd_sof_intel_ipc snd_sof_xtensa_dsp snd_sof_intel_hda_common snd_hda_codec_realtek vfat snd_soc_hdac_hda fat snd_hda_codec_generic kvm snd_sof_intel_hda snd_sof snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core iwlmvm snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_nhlt snd_hda_codec irqbypass snd_hda_core rapl snd_hwdep intel_cstate intel_uncore psmouse pcspkr input_leds e1000e mac80211 i2c_i801 snd_pcm
[ven. oct. 30 11:16:42 2020]  snd_timer thunderbolt libarc4 iwlwifi mei_me intel_lpss_pci rtsx_pci_ms intel_lpss memstick i915 mei idma64 cfg80211 btusb btrtl btbcm btintel uvcvideo bluetooth videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common i2c_algo_bit videodev drm_kms_helper intel_gtt processor_thermal_device ecdh_generic syscopyarea mc ecc sysfillrect ucsi_acpi intel_rapl_common sysimgblt typec_ucsi fb_sys_fops intel_soc_dts_iosf intel_pch_thermal typec wmi thinkpad_acpi tpm_crb nvram ledtrig_audio rfkill snd soundcore battery tpm_tis int3403_thermal tpm_tis_core ac int340x_thermal_zone tpm rng_core evdev int3400_thermal mac_hid acpi_thermal_rel msr sg drm scsi_mod crypto_user acpi_call(OE) agpgart ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid dm_crypt dm_mod rtsx_pci_sdmmc mmc_core serio_raw crct10dif_pclmul atkbd crc32_pclmul libps2 crc32c_intel ghash_clmulni_intel aesni_intel crypto_simd xhci_pci cryptd
[ven. oct. 30 11:16:42 2020]  glue_helper xhci_hcd rtsx_pci i8042 serio
[ven. oct. 30 11:16:42 2020] CPU: 2 PID: 6751 Comm: zsh Tainted: G           OE     5.4.72-1-lts #1
[ven. oct. 30 11:16:42 2020] Hardware name: LENOVO 20N3S37900/20N3S37900, BIOS N2IET91W (1.69 ) 08/12/2020
[ven. oct. 30 11:16:42 2020] RIP: 0010:pci_disable_device+0x9e/0xc0
[ven. oct. 30 11:16:42 2020] Code: 4d 85 e4 75 07 4c 8b a3 b0 00 00 00 48 8d bb b0 00 00 00 e8 c4 16 16 00 4c 89 e2 48 c7 c7 78 d8 d0 8a 48 89 c6 e8 f0 b7 ba ff <0f> 0b e9 79 ff ff ff 48 89 df e8 d3 fe ff ff 80 a3 c9 07 00 00 df
[ven. oct. 30 11:16:42 2020] RSP: 0018:ffffbc45c4cf3e50 EFLAGS: 00010282
[ven. oct. 30 11:16:42 2020] RAX: 0000000000000000 RBX: ffff9d9777ce2000 RCX: 0000000000000000
[ven. oct. 30 11:16:42 2020] RDX: 0000000000000033 RSI: ffffffff8b4dab73 RDI: 0000000000000246
[ven. oct. 30 11:16:42 2020] RBP: 00005602672e8240 R08: 00000358feaf7388 R09: 0000000000000033
[ven. oct. 30 11:16:42 2020] R10: 0000000000001a5f R11: 0000000000000001 R12: ffff9d9777c58900
[ven. oct. 30 11:16:42 2020] R13: ffffbc45c4cf3f10 R14: 00005602672e8240 R15: 0000000000000000
[ven. oct. 30 11:16:42 2020] FS:  00007f72550f7f80(0000) GS:ffff9d977c480000(0000) knlGS:0000000000000000
[ven. oct. 30 11:16:42 2020] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ven. oct. 30 11:16:42 2020] CR2: 00002dce40ad8000 CR3: 0000000864d56005 CR4: 00000000003606e0
[ven. oct. 30 11:16:42 2020] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ven. oct. 30 11:16:42 2020] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ven. oct. 30 11:16:42 2020] Call Trace:
[ven. oct. 30 11:16:42 2020]  bbswitch_off.cold+0xc1/0x1da [bbswitch]
[ven. oct. 30 11:16:42 2020]  bbswitch_proc_write+0xe8/0x140 [bbswitch]
[ven. oct. 30 11:16:42 2020]  proc_reg_write+0x3e/0x80
[ven. oct. 30 11:16:42 2020]  vfs_write+0xb6/0x1a0
[ven. oct. 30 11:16:42 2020]  ksys_write+0x67/0xe0
[ven. oct. 30 11:16:42 2020]  do_syscall_64+0x49/0x90
[ven. oct. 30 11:16:42 2020]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ven. oct. 30 11:16:42 2020] RIP: 0033:0x7f72551e9f67
[ven. oct. 30 11:16:42 2020] Code: 0d 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b7 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 48 89 54 24 18 48 89 74 24
[ven. oct. 30 11:16:42 2020] RSP: 002b:00007ffe8742b8e8 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
[ven. oct. 30 11:16:42 2020] RAX: ffffffffffffffda RBX: 0000000000000004 RCX: 00007f72551e9f67
[ven. oct. 30 11:16:42 2020] RDX: 0000000000000004 RSI: 00005602672e8240 RDI: 0000000000000001
[ven. oct. 30 11:16:42 2020] RBP: 00005602672e8240 R08: 000000000000000a R09: 0000000000000000
[ven. oct. 30 11:16:42 2020] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000004
[ven. oct. 30 11:16:42 2020] R13: 00007f72552bc520 R14: 0000000000000004 R15: 00007f72552bc720
[ven. oct. 30 11:16:42 2020] ---[ end trace 84c1da2e253ae795 ]---

the /proc/acpi/bbswitch still says it's on :/ 

Any idea / way to help to debug this?

Thanks

alex0809, 2020-11-09:
I had the same issue, what fixed it for me was adding the kernel parameter 
`acpi_osi='!Linux-Lenovo-NV-HDMI-Audio'` as described in the [Arch Wiki](https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_T490#Extremely_high_battery_usage_because_of_dedicated_GPU).

My laptop power usage now is under 10W on average, fluctuating between 8-12W. Still seems fairly high, but certainly a lot less than the 15W+ I had before the fix to MX250 power management.

ouafnico, 2020-11-11:
@alex0809  Thanks ! 
I try this method, I still got the issue.
bbswitch still get some kernel errors, and says ON.
but says:
```
[   73.323545] pci 0000:3c:00.0: disabling already-disabled device
```
It seems the card is off anyway:

```
-> # cat /sys/bus/pci/devices/0000:3c:00.0/firmware_node/power_state

D3hot
```

I'm still above 12W of power usage