Bumblebee-Project/bbswitch #221: No suitable _DSM call found.

babaruh, 2022-11-02:
dmesg | grep -C 10 bbswitch: here https://pastebin.com/ARgAJeqQ
```
$ uname -a
Linux archlinux 6.0.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 29 Oct 2022 14:08:39 +0000 x86_64 GNU/Linux
```
```
community/bbswitch-dkms 0.8-550 [installed]
```
Submit your machine information on https://bugs.launchpad.net/bugs/752542 - done
```
acpi_dump_info]$ sudo make load
rmmod acpi_dump_info
rmmod: ERROR: Module acpi_dump_info is not currently loaded
make: [Makefile:15: load] Error 1 (ignored)
insmod acpi_dump_info.ko
insmod: ERROR: could not load module acpi_dump_info.ko: No such file or directory
make: *** [Makefile:16: load] Error 1
```

icegood, 2023-01-09:
the same issue for me:

Jan 09 16:34:46 ice-home kernel: bbswitch: loading out-of-tree module taints kernel.
Jan 09 16:34:46 ice-home kernel: bbswitch: module verification failed: signature and/or required key missing - tainting kernel
Jan 09 16:34:46 ice-home kernel: bbswitch: version 0.8
Jan 09 16:34:46 ice-home kernel: bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.GPP0.PEGP
Jan 09 16:34:46 ice-home kernel: bbswitch: Found discrete VGA device 0000:07:00.0: \_SB_.PCI0.GP17.VGA_
Jan 09 16:34:46 ice-home kernel: bbswitch: failed to evaluate \_SB_.PCI0.GP17.VGA_._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
Jan 09 16:34:46 ice-home kernel: bbswitch: failed to evaluate \_SB_.PCI0.GP17.VGA_._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
Jan 09 16:34:46 ice-home kernel: bbswitch: No suitable _DSM call found.

as i see from topic starter's log he succeeded with 
vga_switcheroo: detected switching method \_SB_.PCI0.GP17.VGA_.ATPX handle

JmZeroQAQ, 2023-09-20:
I'm meeting this issue, have any solutions? :(

I used `pacman -S bbswitch-dkms` to install bbswitch.